Tamandutech/LF_STM32 #18: Adicionar parada por Bluetooth

Matheus-de-Sousa, 2021-11-21:
Usar um comando bluetooth para parar o robô.

raphera, 2021-11-22:
@RizziGama, @Matheus-de-Sousa, no passado notamos algumas desvantagens ao uso do bluetooth/WiFi, pelo consumo de tempo de CPU de seus protocolos.

Uma alternativa da espressif para comunicação, que utiliza a mesma banda do WiFi e Bluetooth de 2.4GHz, é o protocolo ESP-NOW, a carga em cima da CPU é mínima e otimizada, além de que o alcance é superior ao Bluetooth/WiFi.

Tínhamos em desenvolvimento um dashboard que mostrava para nós todos os dados em tempo real, utilizando ESP-NOW, e nos permitia setar alguns parâmetros de PID. Podemos retomar o desenvolvimento desse projeto tornando ele modular para a categoria/equipe já que a maioria de nós utiliza agora o ESP32.

raphera, 2021-11-22:
Abri a seguinte discussão para tratarmos do tema: [Interface para controle/parametrização do robô](https://github.com/orgs/Tamandutech/teams/linefollower/discussions/1).

raphera, 2022-01-11:
O ESP-NOW será utilizado em conjunto com uma dashboard para controle do robô ao invés do Bluetooth.

closed 2022-01-11 by raphera