Tamandutech/LF_STM32 #34: Criar comandos para ler ou substituir o mapeamento inteiro

Matheus-de-Sousa, 2022-06-07:
Criar comandos no Better Console que permita a leitura do mapeamento inteiro com apenas um comando e também desenvolver um comando para alterar todo o mapeamento com só um comando

raphera, 2022-06-07:
Precisamos nos atentar as limitações quanto ao tamanho dos comandos, enviar Strings grandes demais pode ocasionar problemas na transmissão e no processamento do comando. Acredito que a melhor forma seja editar item por item da lista de mapeamento.

raphera, 2022-06-07:
Pode-se implementar o seguintes comandos

- **map_get** : retorna quantidade de mapeamentos
- **map_get X** : retorna o mapeamento no índice X

- **map_delete** : remove todos os mapeamentos
-  **map_delete X**: remove o mapeamento no índice X e desloca os demais

- **map_insert** : insere um mapeamento no final da lista
- **map_insert X** : insere um mapeamento no índice X e desloca os demais

- **map_update X** : edita os dados do mapeamento no índice X

Matheus-de-Sousa, 2022-06-07:
Eu acho a ideia boa, mas eu só tô com dúvida de como ficaria o controle para verificar se houve erro no envio ou recebimento do mapeamento, pois teríamos que verificar comando por comando, ao invés de verificar apenas um comando.

raphera, 2022-06-07:
> Eu acho a ideia boa, mas eu só tô com dúvida de como ficaria o controle para verificar se houve erro no envio ou recebimento do mapeamento, pois teríamos que verificar comando por comando, ao invés de verificar apenas um comando.

Podemos deixar o gateway encarregado de manter a integridade dia dados durante as transmissões.

Matheus-de-Sousa, 2022-06-09:
Criei alguns dos comandos discutidos, porém creio que não precisemos nos preocupar tanto com a ordenação dos registros, pois uma vez que temos os dados do mapeamento é possível utilizar os comandos de ordenação com base na média dos encoders para ordená-lo, além disso, criei comandos para trabalhar com o mapeamento na Ram e na flash para evitar que o mapeamento salvo na flash corrompa.

raphera, 2022-06-10:
> Criei alguns dos comandos discutidos, porém creio que não precisemos nos preocupar tanto com a ordenação dos registros, pois uma vez que temos os dados do mapeamento é possível utilizar os comandos [...]

Gostei da ideia de alterar na RAM e salvar sob demanda na flash, isso poupa ciclos da memória flash e evita mudanças permanentes sem interesse.

closed 2022-06-30 by Matheus-de-Sousa