Tamandutech/LF_STM32 #35: Estudo do código atual

raphera, 2022-06-11:


Pedromone, 2022-06-16:
De primeira estava dando alguns erros de falta de extensões e necessitando assim instalando as mesmas

De segunda, tive alguns problemas de lentidão na leitura dos códigos, e erros nas instalações, precisando assim, um quick start no platformIO

De terceira e gloriosa vez, eu e o matheus ficamos debrulhando o motivo da platformIO não conseguir encontrar seu proprio arquivo, por conta disso, fizemos algumas formatações tanta nas extensões quanto no proprio codigo. Por fim, deu certo.
Com a build sendo rodada e "inicializando/processando" todas as instancias.

Aqui vai algumas imagens da saga 
[image]
[image]
[image]

Matheus imagens abaixo
[image]
[image]

Pedromone, 2022-06-16:
PROCESSO DE BUILDAR TUDO DE UMA VEZ
[image]

Pedromone, 2022-06-16:
:)
[image]

giovannirdias, 2022-06-19:
**O que o robô faz assim que liga?**

No primeiro momento foi isso que entendi ao ler a main_robot.cpp e os arquivos de execução e cabeçalho nos diretórios de serviços e dados. Se identificarem algo de errado ou que tiver que complementar algum ponto sobre a inicialização, avise-me se puder.

Inicialmente cria-se as instâncias referentes aos serviços e os dados dos robôs, com a utilização do serviço LEDsService para indicar a partir dos comandos de cores qual é operação está sendo executada no main_robot.cpp (ESP_LOGD &#8594;  Método de registro).  

```
ESP_LOGD("MAIN", "LED Vermelho");
ledsService->queueCommand(command);
mappingService = MappingService::getInstance("MappingService", 8192, 18);
.........................................................................

ESP_LOGD("MAIN", "LED Laranja");
command.color = LED_COLOR_ORANGE;
ledsService->queueCommand(command);
sensorsService->Start();
```

Após isso, o robô terá seu status colocado no modo parado, aguardando o pressionamento do botão novamente (xQueueReceive(gpio_evt_queue, &num, portMAX_DELAY)) para modificar esse status para linha ou curva, executando a função de mapeamento da pista para identificar as características e os valores dos parâmetros referentes ao mapeamento, PID (velocidade rotacional e translacional) e velocidade do motores.

closed 2022-07-07 by Matheus-de-Sousa