Tamandutech/LF_STM32 #46: Procurar métodos para sintonizar o PID

Matheus-de-Sousa, 2022-07-07:


Matheus-de-Sousa, 2022-07-07:
Utilizar sensores laterais e frontais para sintonizar o PID.

giovannirdias, 2022-07-12:
Busquei algumas referências para métodos de sintonização do PID e apontei algumas considerações em relação a sua implementação. 

- **Descrição da tarefa:** sintonização de um controlador PID é um tarefa de ajuste dos parâmetros desse controlador para obtenção de um sistema com bom desempenho. 
- **Objetivo:** queremos respostas rápidas em transições de estado de pista, robustez à perturbações em alta velocidades em longas retas e diminuição de ações de controle excessivo que limitam a performance dos motores do Braia.

Inicialmente foquei nos métodos clássicos para ajuste dos parâmetros, já que são muito utilizados pela comunidade em sistemas de controle aplicadas na categoria segue-linha. Outra frente seria utilizar métodos adaptativo, que utilizam aspectos de inteligência computacional, algo interessante de se explorar, mas temos que ver a viabilidade e eficiência para competições.

giovannirdias, 2022-07-12:
## Métodos Clássicos (Heurísticos)

- Esses métodos de sintonização não são tão bons para sistemas parcialmente conhecidos e com condições de operação variável.

### Método I  - Tentativa e Erro

- **Descrição:** basicamente é um ajuste manual analisando o comportamento do Braia, assemelhando muito ao que fizemos nessa competição, porém achei uma descrição mais detalhada de como utilizar esse método de maneira sistemática.
<p align="center">
[image]
</p>

- Para análise do comportamento podemos observar as influência que cada parâmetro produz no robô, talvez tenha como usar os dados do dashboard para analisar a curva de saída dos encoders e identificar as características.
1. $K_p$ → aumenta a velocidade da resposta, mas implica em aumento de oscilações a partir de um valor. Quando o carrinho ultrapassa linhas deve-se reduzir o $K_p$, já quando ele está lento ou não conseguir fazer curvas, deve-se aumentar $K_p$.
2. $K_i$ → atingir resposta de estado estável desejada, porém resultando em respostas oscilante em uma maior período. Como $K_i$ é cumulativo, quando está em valores altos empurra o robô para os lados, em algumas aplicações é utilizado para altas velocidades.
3. $K_d$ → amortecer o sinal de saída para referência, mas tornando o sistema sensível a ruídos. Deve-se aumentar $K_d$ no caso de haver oscilações no sistema após $K_p$ atingir o valor supremo.

### Método 2 - Método de Ziegler-Nichols

- Método heurístico para obter resposta de frequência, colocando o sistema no limite de estabilidade, sendo baseado em regras pré-estabelecidas e informadas em tabelas
- Não é um método tão eficiente para rastreamento de setpoint, o que precisamos no caso do seguidor de linha. Facilita a sua utilização caso definimos uma função de transferência (relação entre saída e entrada) da planta, ou seja, modelagem dos motores e da cinemática do segue linha.
- O método consiste em zerar $K_i$ e $K_d$ e $K_p$ é aumentado até que o robô oscile. Quando a oscilação iniciar temos o ganho supremo ou crítico $K_u$ e o último período de  oscilação $T_u$. Depois os PID são ajustados conforme a coluna tabular da tabela.

-> Equação PID:
 <p align="center">
<img style="display: block;-webkit-user-select: none;margin: auto;cursor: zoom-in;background-color: hsl(0, 0%, 90%);transition: background-color 300ms;" src="https://s3.us-west-2.amazonaws.com/secure.notion-static.com/cf621714-665a-4ec3-a6df-9bd4a5d2f533/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&amp;X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220712%2Fus-west-2%2Fs3%2Faws4_request&amp;X-Amz-Date=20220712T160017Z&amp;X-Amz-Expires=86400&amp;X-Amz-Signature=9259b6fd70fbf6fbc0e4f7f5b1c5a4b4c219c3cbec3160688ba1ef26f52a5ed0&amp;X-Amz-SignedHeaders=host&amp;response-content-disposition=filename%20%3D%22Untitled.png%22&amp;x-id=GetObject" width="400" height="200">
</p>
-> Tabela do método:
<p align="center">
<img style="display: block;-webkit-user-select: none;margin: auto;cursor: zoom-in;background-color: hsl(0, 0%, 90%);transition: background-color 300ms;" src="https://s3.us-west-2.amazonaws.com/secure.notion-static.com/158a19c0-a956-49e5-b196-d96ca976f975/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&amp;X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220712%2Fus-west-2%2Fs3%2Faws4_request&amp;X-Amz-Date=20220712T160028Z&amp;X-Amz-Expires=86400&amp;X-Amz-Signature=d94e5f5cdaf31b098fa15ea87d517f366564175db843164a68b4f3396e0758f9&amp;X-Amz-SignedHeaders=host&amp;response-content-disposition=filename%20%3D%22Untitled.png%22&amp;x-id=GetObject" width="400" height="400">
</p>

**IDEIA DE ALGORITMO:** pensei em utilizar os dados de distância do encoder para calcular tempo e o sinal de erro através do setpoint e a saída para assim determinar o período $T_u$ e aplicar as regras da tabela, isso tudo implementado em C++ sob um classe criada chamada PIDAutoTune. O problema é que para determinar $K_u$ temos que realizar voltas nas pistas reais ou em ambientes simulados, neste último caso as possibilidade de simulação estão na outra issue (Mapeamento com IA).
- Têm alguns trabalhos de pessoas que utilizaram esse método no robô seguido de linha, coloquei nas referências caso alguém queira verificar.

### Considerações finais
- Os métodos clássicos são muito rápidos porém exaustivos, tem como aplicá-los, mas como vi nas competições a pista quase sempre está ocupada por outras equipes, então seguir arrisca esse métodos pode ser muitas vezes inviáveis. Uma alternativa para contornar esse problema seria a simulação, mas ter esse ambiente virtual eficiente vai levar um tempo longo de preparação, para que temos condições ótimas para validação desse parâmetros e com isso aplicar na pista real, uma outra alternativa é os métodos adaptativos.

giovannirdias, 2022-07-12:
### Métodos alternativos
- Encontrei um cara que fazia um seguidor de linha utilizando a biblioteca QTRSensors.h, a qual utilizamos também, ele utiliza os valores informados pelo sensores para definir os parâmetros do PID.
- Basicamente ele diz que os sensores de reflectância retornam valores entre 0 a $1000  \times (NumSENSORES-1)$  pensando em medidas analógicas, assim como o Braia tem 8 sensores frontais, o valor de alcance seria de 0 até 7000. 
- O objetivo é ter uma leitura de sensor em 3500 (meio da pista),  com isso para definir $K_p$ inicialmente pegaria o maior erro possível e a velocidade máxima permissível, ficando assim $K_p = \frac{\text{velocidade desejada}}{erro}$, depois definimos $K_D$ como 10 vezes o valor de K_p.

<p align="center">
<img style="display: block;-webkit-user-select: none;margin: auto;cursor: zoom-in;background-color: hsl(0, 0%, 90%);transition: background-color 300ms;" src="https://s3.us-west-2.amazonaws.com/secure.notion-static.com/6b914be1-bcc2-49b0-a9de-20ad86b4e6ca/IR-Sensor-Array-Motor-Speeds.webp?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&amp;X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220712%2Fus-west-2%2Fs3%2Faws4_request&amp;X-Amz-Date=20220712T163223Z&amp;X-Amz-Expires=86400&amp;X-Amz-Signature=a662d05474ffb494e9dd7a1721fe1cbbf7ae16298c9810a6b4782b3e31b97a50&amp;X-Amz-SignedHeaders=host&amp;response-content-disposition=filename%20%3D%22IR-Sensor-Array-Motor-Speeds.webp%22&amp;x-id=GetObject" width="300" height="200">
</p>

- É um método experimental que podemos testar, mas talvez seja restrito ao formato do robô dele e as pistas específica que ele utilizou.

**Referência**
1. https://robotresearchlab.com/2019/02/16/pid-line-follower-tuning/

giovannirdias, 2022-07-12:
# Métodos Adaptativos

- Esses métodos são de auto-ajuste do PID baseados na campo de _machine learning_. Eles permitem determinar os parâmetros mesmo que não conhecemos o sistema por completo e suas condições de operação são variáveis, o que acontece no caso do Braia.
- Procurei algumas abordagens utilizando inteligência computacional, destacando as limitações para nosso projeto e maneiras da adaptar esses algoritmos para um seguidor de linha.

## Redes Neurais (NN)
- **Ideia:** utilizar a saída de uma rede neural para auto-sintonizar os ganhos $K_p$, $K_i$ e $K_d$, utilizando como dados de entrada os dados de setPoint e a função erro, como mostrado no diagrama de blocos abaixo:

<p align="center">
    <img src="https://s3.us-west-2.amazonaws.com/secure.notion-static.com/62e0e54a-f4b4-4cf4-8404-21277a8b2cb8/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&amp;X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220712%2Fus-west-2%2Fs3%2Faws4_request&amp;X-Amz-Date=20220712T224813Z&amp;X-Amz-Expires=86400&amp;X-Amz-Signature=ab53e9dd3e70188a36ab15cb3d27e094ac33d480ed3f98043a5a6cb2a7d49491&amp;X-Amz-SignedHeaders=host&amp;response-content-disposition=filename%20%3D%22Untitled.png%22&amp;x-id=GetObject" alt="https://s3.us-west-2.amazonaws.com/secure.notion-static.com/62e0e54a-f4b4-4cf4-8404-21277a8b2cb8/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&amp;X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220712%2Fus-west-2%2Fs3%2Faws4_request&amp;X-Amz-Date=20220712T224813Z&amp;X-Amz-Expires=86400&amp;X-Amz-Signature=ab53e9dd3e70188a36ab15cb3d27e094ac33d480ed3f98043a5a6cb2a7d49491&amp;X-Amz-SignedHeaders=host&amp;response-content-disposition=filename%20%3D%22Untitled.png%22&amp;x-id=GetObject" class="shrinkToFit transparent" width="500" height="300">
</p>

- Procurando referências, encontrei alguns _papers_ de pessoas que aplicaram alguns modelos de NN em seguidor de linha, na maioria ou utilizando simuladores ou aplicando o algoritmo NN online.

- **Vantagens:** extremamente adaptativos (algo interessante para pista com diversos tipos de curvas), habilidade de aprendizado e capacidade de mapeamento não-linear.
- **Desvantagens:** custos computacionais dependendo das especificações do microcontrolador, ou seja, utilizado o algoritmo em _real-time_ e trabalhando online temos desafios em relação a sistemas extremamente não-lineares [1].

### IDEIA
- Na minha opinão, podemos abordar de três maneiras a implementação de _auto-tuning_  do controlador PID:
1. NN online: podemos utilizar um procedimento simultâneo, pegar os dados gerados na volta e aplicá-los nas entradas para conseguir otimizar os parâmetros no mesmo tempo que o carrinho está na pista. Outro procedimento seria não-simultâneo, onde poderiamos importar os dados do dashboard referente ao mapeamento e acrescentar talvez algumas voltas com parâmetros arbitrárias para ser os _inputs_ e gerar os parâmetros otimizados que devem ser apresentados no dataPid.
2. NN no ESP32: nesse caso provavelmente teriamos que ter mais um microcontrolador no Braia, porque teríamos muitas computações sendo feitas no processador, afetando a sua velocidade e consequentemente a percepção de linhas do Braia. Assim, nessa abordagem utilizáriamos as redes neurais simultaneamente com o seguidor de linha, otimizando os parâmetros de acordo com o estado da pista; ou não simultâneo, aplicando sob os dados de mapeamento (atribuindo inicialmente parâmetros arbitrários) ou armazenamento os dados de uma volta e ao finalizar estabelecer os parâmetros ótimos.

- Em ambas abordagens a saída da rede neural considerando apenas os parâmetros lineares seria:
$u(t) = K_{p,NN} e(t) + K_{i, NN} \int {e dt} + K_{d, NN} \frac{d e(t)}{dt}$

- No caso de limitação computacional podemos talvez tentar redes menores, como ADELAINE que só tem uma única camada. Outro modelo seria utilizar _backpropagation networks_, só que seria necessário um algoritmo de aprendizagem secundário para estabelecer os ganhos PID.

- **Observação:** o TensorFlow Lite funciona no ESP32, abrangendo implementações em C++ e Python, o que facilitaria a utilização de bibliotecas para construir as redes neurais. Além disso, o ESP32 tem como ser programado na linguagem Python através do _MicroPython module_, mas temos que lembrar que  o Python não é uma linguagem com alta performance, então não sei como compartaria-se o sistema em _real-time_ utilizando essa linguagem. 

### Referências
1. [https://www.researchgate.net/publication/324537461_Tuning_PID_Controller_by_Neural_Network_for_Robot_Manipulator_Trajectory_Tracking](url)
2. [https://github.com/antevis/CarND_T2_P4_PID](url)

## Algoritmos Genéticos (GA)
- Abordagens interessantes, porém impraticáveis fora de um ambiente simulado. Na issue #44  destaco como poderiamos fazer a simulação, limitações físicas e os desafios com os simuladores e criação de pistas.
- Nessa referência [2], tem um exemplo de simulação de otimização do controlador PID utilizando GA a partir do ROS e o V-REP, ambiente virtual da _Coppeliasim_.

### Refereências
1. [https://www.researchgate.net/publication/296077413_Tuning_PID_Controllers_Using_Artificial_Intelligence_Techniques_Applied_to_DC-Motor_and_AVR_System](url)
2. [https://github.com/MatheusBorgesKamla/Seguidor_de_Linha_Evolutivo](url)

## Q-Learning (Aprendizado por Reforço - RL)

- Algoritmo iterativo que pode ser utilizado para otimização de controladores em sistemas robóticos.
[... cut ...] 2-4-Implementation.png%22&amp;x-id=GetObject" class="shrinkToFit" width="725" height="394">
</p>

#### Referências
1. [https://www.researchgate.net/publication/322930011_Q-Learning_for_Adaptive_PID_Control_of_a_Line_Follower_Mobile_Robot](url)
2. [https://www.youtube.com/watch?v=NrLjs8JVGNU](url)
3. [http://vigir.missouri.edu/~gdesouza/Research/Conference_CDs/IEEE_WCCI_2020/IJCNN/Papers/N-21062.pdf](url)
4. [https://www.hackster.io/aslamahrahiman/deep-reinforcement-learning-on-esp32-843928](url)

Matheus-de-Sousa, 2022-07-15:
### Minha Proposta
- Ao analisar os métodos propostos creio que o mais interessante é o método com Redes neurais online não simultâneo em que escolhemos alguns valores arbitrários para as voltas e depois exportamos esses dados pelo dashboard para que sejam processados pela rede neural, com isso poderíamos começar a fazer alguns testes e no futuro utilizar o método simultâneo em que precisaríamos integrar o robô ou o dashboard com uma aplicação com rede neural através de conexão sem fio, poderíamos utilizar o protocolo websocket que já utilizamos no dashboard para fazer essa integração, mas para começar acredito que seja mais fácil o método não simultâneo.
### Etapas para implementação
1. Especificação dos dados da pista e do robô que serão necessários e em que formato eles devem estar.
2. Preparar as estrutura de dados e métodos do robô para armazenar e enviar os dados necessários para o dashboard.
3. Utilizar o dashboard para receber e salvar esses dados em um arquivo para processamento pela rede neural.
4. Criação do algoritmo que implemente a rede neural no processamento dos dados contidos no arquivo e que retorne os valores otimizados que podem ser alterados manualmente pelo dashboard.

raphera, 2022-07-16:
@mCaburlao se juntará a atividade;
@Pedromone buscará materiais complementares.

closed 2025-05-05 by samuelc254