Tamandutech/LF_STM32 #53: K-Learning para otimizar o PID.

raphera, 2022-09-07:
Criar um algoritmo que otimiza um PID relativamente bom já determinado:

mCaburlao, 2022-10-01:
Como não existem muitos artigos testando o Reinforcement Learning, é difícil definir quais os parâmetros ideias para o funcionamento do algoritmo, considerando que não podemos deixar o robô rodando por um tempo indeterminado. Por causa disso, seria ideal testar o algoritmo primeiramente em uma simulação.

Eu encontrei uma simulação de um robô segue-linha no programa Webots, além de um framework próprio do programa para Q-Learning. Também é possível transferir um modelo 3D próprio ao programa, mas só encontrei como importar modelos do TinkerCAD por enquanto. Diferente do CoppeliaSim, nesse programa parece ser mais fácil importar uma pista, se ela também for montada no TinkerCAD.

Fontes do modelo e das bibliotecas:
 - Teste de um robô real VS modelo: https://www.youtube.com/watch?v=tanIEUHYbxE&list=PLhwo1YHRc6DaekhQPNaz-T4s-m3rnAxNt&index=9&ab_channel=DrakerDGRobotics
 - Construção de uma pista no WB: https://youtu.be/XUSCD7aYtQ8
 - Simulação de um segue linha: https://github.com/DrakerDG/Line-Follower-Robot-V4
 - Deepbots framework: https://github.com/aidudezzz/deepbots-tutorials

giovannirdias, 2022-10-01:
# Fase 1 - Construção do Algoritmo de _Reinforcement Learning_ (RL)

Começei a desenvolver o modelo de R utilizando alguns papers de referência, especificamente o modelo _soft actor-critic_ que é um modelo de otimização de política estocástica do tipo _off-policy_ utilizando uma dupla função Q incorporado com redes neurais profundas em estágios denominados ator e crítico. 

Como estou me baseando nas informações dos paper, possivelmente falta ainda alguma implementação das equações que eles mostram, então o código que escrevi está ainda em estágio básico, não considerei ainda a parte de comunicação com robô e com relação a possibilidade de escrever-lo com C++, seria necessário utilizar o tensorflowlite, porém acho q a parte de treinamento em real-time não seja efetivo, já rodar na própria ESP é algo computacionalmente custoso, no paper eles utilizam um NVIDA Jetson para isso.

Por ser um esboço não criei a branch aqui, mas deixei em um repositório pessoal público para vcs conseguirem ver e modificarem algo se achar necessário, o link é esse abaixo:
https://github.com/giovannirdias/SAC-PID

## Descrição do Modelo

Nessa área de aprendizado por reforço muitos nomes são recorrentes para os mais diversos tipos de modelos, então acho que seria interessante resumir um pouco esses termos para facilitar a inserção de alguém interessado neste projeto e também justificar minha escolha pelo modelo que disse anteriormente.

- Reinforcement Learning *RL) x Teoria de Controle (PID, Avanço-Atraso, ...)
O RL é uma subárea do campo de inteligência artificial, especificamente relacionado a aprendizado de máquina, responsável pela construção de sistemas que utilizam dados para manipulação futura em geral de sistemas dinâmicos, algo intimamente ligado à teoria de controle, principalmente com teoria de controle ótimo que envolve processo de decisão de Markov. Desta forma, podemos estabelecer uma correlação com os conceitos dos campos, que divergem mais pelas técnicas para manipulação da informação até atingir um estado quase ótimo.

- Objetivo utilizando RL
Queremos que algoritmo estabeleça uma relação entre os estados e ações de uma agente, o que demonina-se como política, para assim tomar decisões adequadas para permitir a maximização da recompensa dada ao agente interagindo com o ambiente ao qual está inserido. O algoritmo funciona tanto com dados de treinamento quanto sem eles, porém nesse casso ele deve aprender por experiência própria através de um 'tentativa e erro inteligente'.

- Principais termos da RL
1. Estados: uma configuração instantânea ou futura de um agente com relação ao ambiente, por exemplo no nosso caso os estados são as posições do braia sobre a pista;
2.  Ação: são um conjunto de movimentos que influência no estado do agente, geralmente discretizadas porém há novos modelos que trabalham com valores contínuos, algo fundamental para nós, já que nossas ações são os ganhos de PID que desejamos otimizar, são eles que possuem influência completa no controle da posição do robô.
3. Recompensa: um feedback utilizado para instruir o robô a priorizar ações que levem em direção aos objetivos exigidos. Assim, estabelece regras que governam as ações do agente durante os estados que ele for se encontrando durante o seu trabalho.
4. Política: equivale a uma função ou estratégia utilizada para determinação das futuras ações de acordo com o estado atual, ou seja, ela é responsável pelo mapeamento dos estados para ações, convergindo para um possível conjunto de ações que impliquem em máxima recompensa.
5. Value: estipulação do retorno ao longo prazo de acordo com o estado atual e a política vigente, assim é utilizadas para observar as viabilidades dos caminhos seguidos.
6. Q-value: avaliação de longo prazo das ações durante uma política. É responsável pelo mapeamento dos conjuntos de ações e estados para um recompensa.

- Porque o _soft actor-critic_?
Bem, nosso problema está inserido em condições de ações dentro do espaço contínuo, já que os ganhos do PID são do tipo float. Desta forma, torna-se inviável a utilização do algoritmo tradicional de Q-learning.
Buscando mais informações sobre a problema de sintonização de PID, encontrei dois artigos que utilizam uma versão de RL baseado na regularização das ações através de entropia (aleatoridade de uma variável aleatória). Basicamente, esses modelos treinam as políticas para maximização a troca entre retorno por recompensa e a entropia, algo relacionado com trade-off de _exploration e explotation_, permitindo o acelerando do aprendizado mais tarde durante a exploração para evitar convergências em mínimos ou máximos locais.

- Soft actor-critic função no projeto
q

giovannirdias, 2022-10-23:
# Fase 2 - Simulação do Seguidor de Linha com PID Adaptativo

- Para implementar os testes com sintonizador PID, vamos seguir o que a @mCaburlao comentou acima. Através do simulador WeBots iremos executar o processo de treinamento para armazenar valores de pesos da rede neural, que serão utilizado no segue-linha físico a partir de um bloco paralelo executado simultaneamente a partir do _TensorFlowLite_ que tem aporte no ESP32.

## Planejamento
1. [ ] Aprendendo as funcionalidades do simulador e realizando as modificações do modelo exemplo, tentando incluir neste projeto uma rede neural para estabelecer os parâmetros K a partir de algumas _epochs_ (voltas nas pistas) com ganhos inciais ótimos, porém com algumas oscilações, velocidade não ideal e tempo de percurso alto.
**Prazo estimado:** Acredito que até 30/10 completo essa _task_.

2. [ ] Simulação utilizando como modelo o Braia, adaptando algumas partes do código exemplo para assemelhar com o código real do Braia. 
**Observação:** Estou tendo dificuldade da modelagem do ambiente para incluir o robô no simulador. Estou tendo utilizar a descrição da modelagem do fusion a partir de .obj como algumas referências apresentam, mas o robô não está aparecendo em um mundo estático que criei para testes. Estou pesquisando algumas alternativas para representação do modelo.
**Prazo estimado:** Talvez até dia 06/11 essa _task_ esteja completa.

3. [ ] **Treinamento do modelo:**  Após treinar o robô dentro do simulador por uma considerável quantidade de instâncias (ainda não sei dizer quantas seriam necessárias), teremos uma rede neural com pesos treinado para levar saídas que proporcionam um erro quadrático minimizado (sinal de erro utilizado no controlador PID) para pistas de testes. Na sequência utilizaremos o conversor do modelo para modo do [TensorFlow Lite](https://www.tensorflow.org/lite/microcontrollers/build_convert#model_conversion) para permite a geração do modelo que possa ser aplicado dentro do microcontrolador.
**Observações:** Há algumas restrições da utilização de funções e módulos do TensorFlow nos microcontroladores, então o modelo deve possuir algumas adaptações para execução correta no ESP.
**Prazo estimado:** 27/11.

4. [ ] Adaptações no código do Braia, principalmente nos serviços de PID, para permitir a utilização da rede neural na atualização dos parâmetros proporcional, integrativo e derivativo, e avaliação da efetividade do projeto no desempenho dele em um pista distinta ao do treinamento.
**Observação:** Como nunca mexi com TensorFlow em microcontrolador, provavelmente haverá algumas barreiras durante está tarefa, implicando em maior duração.
**Prazo estimado:** Em Janeiro de 2023 acredito que o projeto esteja finalizado para avaliar a sua efetividade.